Alarm-Siren/arduino-kicad-library #1: How to join GND1, GND2 and GND3 together?

njh, 2017-11-22:
Thank you very much for creating arduino-kicad-library. It looks like just what I need.

Having created a schematic with your Uno Shield, I am having trouble with the Electric Rules Checker. There seem to be a couple of problems:
* It doesn't seem to like that I am joining the GND1, GND2 and GND3 nets together
* I am trying to take the 12v power supply connected to VIN and connect it to some 2N7002 MOSFETs, however it doesn't think the 12V power net is driven

I have found Kicad to be really difficult to get started with. If you are able to provide any help, that would be fantastic!

[image]

Alarm-Siren, 2017-11-23:
I do not have time to look at this in depth right now, I shall get back to you either this evening or tomorrow, depending on how things pan out. In the mean time, it would be helpful if you could give me your whole schematic - either via github link,  attach it to the issue, or even just a picture of the whole thing would do so long as its legible -  so I can see what kinds of things are connected to your +12V rail.

njh, 2017-11-23:
Thanks! I will get it on GitHub tonight...

njh, 2017-11-23:
Repo is here: https://github.com/njh/relay-driver-shield

PDF attached in case of problems.

[relay-driver-shield.pdf](https://github.com/Alarm-Siren/arduino-kicad-library/files/1500292/relay-driver-shield.pdf)

Alarm-Siren, 2017-11-24:
The +12V rail problem is what I thought it was. The default ERC rules require that a Power Input type pin be connected to a Power Output type pin. Your +12V rail is only connected to the outputs of Diodes and the screw terminal, both of which are Passive pins. So, to the ERC, the +12V rail is not driven when it should be - hence error. There is a special component called PWR_FLAG which you can use in this instance; it has a single pin which you attach to the relevant net and this pin is set to Power Output - the ERC now sees a Power Input being driven by a Power Output, and is happy. The PWR_FLAG component is special in that it doesn't appear in CvPcb so you don't have to worry about it cluttering up your PCB design, its purely used as a schematic flag to let the ERC know the net is OK.

---

For the GND problem, this comes down to the fact that on Arduino platform its difficult to really define what the power pins do - if you're powering it from a USB, then the GND , +3.3V and +5V pins are Power Outputs, but if you're instead powering it from your shield then GND and +5V are Power Inputs (or, in the case of your specific schematic, +12V and GND are power inputs and +5V is and output - confused yet?) I had to make a decision which to use for the library and I decided that powering the shield from the Arduino was probably the more normal use case, but you're doing the other way around. You can safely ignore the error, given that you know that what you've done is indeed correct - as a programmer yourself I'm sure you're aware that automated error reporting tools aren't always right, and KiCad's ERC is janky as heck. If you want to get rid of the error, you'll need to edit the schematic component in Library Editor to change the GND pins' electrical types from Power Output to Power Input. You'll also then need to put a PWR_FLAG on it, because there'd then be nothing telling ERC that the net is being driven....

There's various other things I could change, like have only one GND pin which maps to all the physical pins, but then you would get DRC errors in the PCB layout for not connecting all of the GND net pins together (if you didn't want to connect them all together for whatever reason), so I think leaving them separate is the more flexible solution.

---

I hope that is all helpful; if you have any more questions do not hesitate to ask. If I can think of a solution to the GND pin problem that I think is superior to the current situation I will update the library.

njh, 2017-11-24:
Thank you very much for your detailed response.

I am actually using VIN as a Power Out on my shield. I am taking the 12v supply that I have plugged into the barrel connector on the Arduino Uno. The screw terminals are for connecting to my relays.

As you say, others might be doing the opposite, so there isn't any good answer. It is a shame you can't override pin options in eeschema - instead of having to edit the component in the Library Editor. But adding a 'PWR_FLAG' next to the VIN pin solves that warning.

I found this answer on Stack Exchange about multiple ground pins:
https://electronics.stackexchange.com/questions/132033/kicad-pin-assigment

I don't completely understand the answer - I am not sure how eeschema knows that all the ground pins (including the invisible ones) are part of the same net. I guess I don't really need to connect all the ground pins together - it just seems sensible.

Alarm-Siren, 2017-11-24:
Yes, the stacked pins approach would help, but it wouldn't solve all the problems and, as previously mentioned, it would mean that in PcbNew it would think you need to connect all the pins together as they'd share a net, which you might now want to do for whatever reason.

As an example, here's a screenshot of an Arduino Due based shield I made (and indeed, this project was the entire reason I made this library in the first place!)

[image]

The four pins circled in blue are 5V pins. Note I have not connected them all together, as its unnecessary and would have made the routing even more challenging than it already was. If I had used the stacked pins method, PcbNew would think they all have the same net and the DRC checker would complain they're not connected together - even though it is deliberate.

One thing I could do is change all the power pin's types to Power Input; then it won't ever complain about you joining them together; but you would have to remember to connect a PWR_FLAG to them if there's no other component connected to the net that is of Power Output type.

njh, 2017-11-25:
I have decided to remove the connection between GND1, GND2 and GND3.

[image]

No more Electrical Rule Check errors!

And successfully created a simple single-layer PCB:

[image]

Thanks for your help!

nick.

closed 2017-11-25 by njh

Alarm-Siren, 2017-11-28:
I am glad you were able to make your PCB, and I am glad to have been able to help you on your way!

FYI: I have just pushed a new version of the repo which will allow you to join the power pins together if you want to, at the cost of adding a PWR_FLAG. I also found and fixed an issue with the Reset pins.

njh, 2017-11-28:
Thanks!

I just tried updating to version 1.1, however I am not sure how to get eeschema to use the new version instead of the old version in relay-driver-shield-rescue.lib / relay-driver-shield-cache.lib ?

Any tips?

Alarm-Siren, 2017-11-28:
I think you can just delete them and it'll find it in the original .lib file instead. But take a copy of them first just in case. If that doesn't work I'll do some experimenting and see what I come up with.