Alarm-Siren/arduino-kicad-library #4: New Boards

Alarm-Siren, 2018-01-09:
Add parts for the Arduino Nano & Mini as Sockets. Need to investigate these parts, find physical dimensions and pinouts etc.

vikramambrose, 2018-01-22:
When using a shield, how does one align the pin headers in the 3dviewer to be exactly over the holes from the arduino footprint?

Alarm-Siren, 2018-01-22:
I'm sorry, but I do not know the answer to your question for certain as I have never used the 3D viewer myself.
If you have a shield footprint in your PCB design, along with my footprint (which presumably you do in order to get both to show up in the 3D viewer), you can make fine adjustments to the position of one or both by hovering over the component and pressing 'E' to bring up the Footprint Properties dialogue, and adjusting the position X/Y in the bottom left corner of the dialogue.
If you just want to check alignment of pins, I've always found it simplest to print the circuit on a piece of paper and physically place the components on the printout.
Further, just for future reference, you should have created a new issue rather than commenting on this unrelated one.

Alarm-Siren, 2018-02-05:
See #6 and #7 .

closed 2018-02-05 by Alarm-Siren