Alarm-Siren/arduino-kicad-library #23: New Board: Arduino Pro Mini

Alarm-Siren, 2018-02-16:


Alarm-Siren, 2018-02-17:
Board is retired, and there seems to be three different pinouts (relating to the A6 and A7 pins) depending on which specific time period and/or manufacturer made it. Unless someone specifically asks for it, I think its just not worth the effort.

Alarm-Siren, 2019-03-11:
Welp, @fow0ryl asked for it ... so I guess I'd better find a way to make it work

fow0ryl, 2019-03-13:
Ok, I have measured some of my clones.
The outline dimesions differ from piece to piece a little bit, but the pins are ok.

Hope my sketch contains all you need. All values in inches ..
[image]

Alarm-Siren, 2019-03-18:
Thank you very much, I think that should be enough at least to cover your variant. I'll try to get a prototype up for the weekend which you can verify against your boards?

Alarm-Siren, 2019-04-02:
How far are the pins from the edge of the board?

Alarm-Siren, 2019-04-02:
I've pushed a beta version of the Pro Mini Socket symbol and footprint into the 2.0.0 branch for @fow0ryl to review.

As per my previous note on the subject, there are several different variants of the Pro Mini. For example, the one pictured on Arduino's website (https://store.arduino.cc/arduino-pro-mini) has a different pin layout to the one @fow0ryl has given me specs for: Two of the Ax pins are in different places, it appears to be missing a GND pin, and the serial programming header pins are reversed.

I think its worth trying to track down as many variants as possible, and trying to come up with a systemic way of identifying them so users can select the right one within this library.

fow0ryl, 2019-05-19:
Hello,
first of all. Thank you very much for your assistance.
Since I get a Pro Mini with original Arduino layout. I decided to create 2 footprints by myself, based on your work.
The "clone" is mostly the version you made for me. The other one is for the orignal layout.
Feel free to integrate them into your library if you like them.

Henning

[Arduino_Pro_Mini_Clone_Socket.kicad_mod.zip](https://github.com/Alarm-Siren/arduino-kicad-library/files/3195015/Arduino_Pro_Mini_Clone_Socket.kicad_mod.zip)
[Arduino_Pro_Mini_Socket.kicad_mod.zip](https://github.com/Alarm-Siren/arduino-kicad-library/files/3195016/Arduino_Pro_Mini_Socket.kicad_mod.zip)

Alarm-Siren, 2022-10-30:
Thank you, I have integrated your work into the new 2.0.0 branch.

commit 1f7bec05 (2022-10-30): Close #23, Close #25
closed 2022-10-30 by Alarm-Siren