Alarm-Siren/arduino-kicad-library #30: Arduino Mega 3.3V power pin is an input instead of output

pintert3, 2022-02-06:
I thought the Arduino 3.3V pin serves as a 3.3V output from the on board regulator. If I make a circuit and try to get a 3.3V out, the DRC keeps popping up an error claiming that I've not connected an output to the input. I don't know if it's Kicad that could have the issue, but that's where I'm at

pintert3, 2022-02-06:
Nevermind, I read the docs again and I understand the decisions you made. Although I'm still wondering why the 3.3V wasn't made an output, knowing how it's never used as an input

Alarm-Siren, 2022-02-18:
Yes, you are right, there is no reason for the 3.3V pin to be a Power Input on the 5V boards, as on those boards the 3.3V is supplied from a local regulator so will always be an output.

On 3.3V-based boards, I believe the Power Input compromise my still be relevant.

Alarm-Siren, 2022-10-30:
I have changed the 3.3V pin to Power Output for the following 5V-based boards:

- Leonardo
- Mega 2560
- Micro
- Nano
- Uno

I have decided to leave the 3.3V-based boards as they are because, on those boards, you can power the microcontroller directly on the 3.3V pin, bypassing the on-board regulator. If you wanted to use this configuration, then having the pin be a Power Output would create ERC errors.

commit c5ffbfdb (2022-10-30): Close #30

Updated 5V-based boards so that their 3.3V pins are Power Output instead of Power Input.
closed 2022-10-30 by Alarm-Siren