Alarm-Siren/arduino-kicad-library #37: New Board: Arduino MKR NB 1500

Alarm-Siren, 2022-10-30:


Alarm-Siren, 2022-12-11:
Pinout and mechanical dimensions are identical to the MKR FOX 1200.

G0rocks, 2025-02-24:
Hello, I'm using this board for a project and I note that it does not have the following:

- [ ] SIM card slot
- [ ] Antenna connector
- [ ] USB micro connector
- [ ] Battery connector
- [ ] UART (I think it's a uart, I'm not so sure)

How can I add these to the board in this library?
Thank you :)

Alarm-Siren, 2025-02-24:
Hi @G0rocks 

I think you may be confused about what these symbols and footprints are for. They are not re-implementations of the actual Arduino modules themselves, rather they are footprints to allow you to plug an Arduino module into your own custom PCB. Given this fact there is no reason why any of the components you listed above would be included in the library, as they do not affect the outline of the footprint (well, except for the USB micro, which is actually present in the footprint insofar as it affects the outline).

G0rocks, 2025-03-25:
Hello
Thanks for the response.
Yes I read the description of the repo to be
"KiCad Symbol & Footprint Library for Arduino Modules (Shields, Sockets and Tiles) " which I understood as that they would be the actual arduino module themselves.
Now I'm curious though since you say these "... are not re-implementations...", does that mean that somewhere online I could find an arduino module for kicad which would include the things I pointed out? If so, where do I find that module? Thank you!

G0rocks, 2025-03-31:
Thank you!
Would it be an idea to add the boards to this library as a board, not a socket?

commit 5e0b6b77 (2022-12-11): Add new board: Arduino MKR NB 1500

Closes #37