Alarm-Siren/arduino-kicad-library #81: Nicla Sense Me not available on KiCad7

ayaghini, 2023-03-12:
Hello,

Nicla Sense Me is missing from the library. I am using KiCad 7 and followed the instruction provided to add the Arduino package to my project.

Alarm-Siren, 2023-03-12:
Hi ayaghini,

Unfortunately the KiCad team have been ignoring the merge request for v4 into the KiCad PCM repo for the last six days, which is why you can't see v4 in the KiCad official repo list yet. As soon as the KiCad team get around to actioning the merge request, you'll be able to download as per the instructions. (see the [merge request here](https://gitlab.com/kicad/addons/metadata/-/merge_requests/186) - if you happen to have a GitLab account, please feel free to give them a kick on my behalf!)

However, you can install v4 (which includes the Nicla Sense ME) manually: download the [v4.0.0 PCM package](https://github.com/Alarm-Siren/arduino-kicad-library/releases/download/v4.0.0/arduino-kicad-library-4.0.0-pcm.zip), and then open it using the "Install from File..." button at the bottom left of the Package and Content Manager (PCM) window. Voila.

I apologise for the inconvenience.

Alarm-Siren, 2023-03-12:
I will leave this issue open until KiCad get around to merging v4 into the PCM repo. Hopefully the information here will be useful to others in the interim.

Alarm-Siren, 2023-03-12:
v4 has now been merged into the KiCad PCM Repo, so you should now be able to update to v4 using the PCM. Again, I apologise for the confusion and inconvenience.

ayaghini, 2023-03-12:
Hi, thank you so much for your professional follow up ,really appreciate your devotion to the community.

Ali

closed 2023-03-13 by Alarm-Siren