Alarm-Siren/arduino-kicad-library #84: In the Pro_Mini_Clone_Socket symbol, MISO and MOSI are backwards

DeflateAwning, 2023-04-05:
Well that was shitty, glad I figured out why it wasn't working though.

Currently, the GPIO pins in the Pro_Mini_Clone_Socket are:
* D11_MISO
* D12_MOSI

Current picture:
[image]

It should be the opposite:
* D11_MOSI
* D12_MISO

Someone should check that these are correct in the other schematic symbols too! Big big mistake.

Alarm-Siren, 2023-04-05:
Thank you for the bug report, you are indeed correct.
I have checked all the symbols for incorrect MISO/MOSI assignment and only the Arduino Pro Mini and Clone Pro Mini are affected.

closed 2023-04-05 by Alarm-Siren

Alarm-Siren, 2023-04-05:
Version 4.0.1 addresses this issue, and I have submitted it to KiCad for inclusion in the PCM (see merge request here: https://gitlab.com/kicad/addons/metadata/-/merge_requests/203). Unfortunately I have no control over how long it will take the KiCad team to review and approve the new release - last time it took six days.

In the mean time you can manually install the updated library using the PCM package file here: https://github.com/Alarm-Siren/arduino-kicad-library/releases/tag/v4.0.1

DeflateAwning, 2023-04-06:
Thank you for fixing this so quickly

commit 792ea0a7 (2023-04-05): Close #84

On the Pro Mini (Arduino and Clone) the MOSI and MISO labels were incorrectly reversed on pins D11 and D12, this commit resolves this issue.