Alarm-Siren/arduino-kicad-library #85: New Module: Arduino MEGA 2560 Pro (also sold as Mega 2560 Embed)

BartoilPerso, 2023-05-12:
Just wondering if this was a board on your radar to add to the library - it's not an official board, but a clone, and potentially a really useful one (I just got several of these in the mail last night)

https://www.pcboard.ca/mega-2560-pro and https://circuit.rocks/product:2657 are two links to the product

Similar specs to the Mega 2560, in a much smaller form factor. 

I'm no good at creating items for a library, but I can certainly attempt to help if needed
Thanks!

Alarm-Siren, 2023-05-13:
I don't normally cover unofficial "arduino compatible" boards unless someone specifically asks for them, but as you've done so I see no reason why not! Unfortunately there's no proper datasheet at either of the links you sent, so unless you can give me the proper mechanical dimensions I can't help (I can probably guess from the picture, but I wouldn't be comfortable publishing a guess).

As you say you've got some of these modules in hand, if you're able to take some measurements that'd be grand. See issue #23 for a previous example. I would need to know the dimensions of the module, the distance between pins in both axes in each group, and the distance of at least one reference pin within each group from the edges of the module. The location and diameter of the two mounting holes, and the position and rough size of the USB connector would also be grand... does the USB stick out from the edge of the board? I can't tell from the pictures.

BartoilPerso, 2023-05-13:
Here's what I've got for measurements for you. As you can tell by the silkscreen, I'm not sure I've got some of the best clones out there, but from what I've read in a few spots they are all very close in size

Please let me know if you need any more details. I'll see if I can find a good pinout diagram, or perhaps make one myself, as the silkscreen is hard to read in a few spots

Alarm-Siren, 2023-05-13:
Yeah, it looks like the board tolerances at the cut edges are fairly large. you've got 60 thou at the top and 50 thou at the bottom. I think for the purposes of the footprint I'll assume it's meant to be 60 thou all the way around: provided the pins themselves are all in the right places relative to each-other, a slightly oversize silkscreen outline won't do any harm.

Did you get the diameters of the mounting holes? and how deep (left-right axis on that picture) is the USB connector?

BartoilPerso, 2023-05-13:
Mounting holes are both 0.130", and the USB connector is flush with the edge of the board. The two inner rows are 1.20" between centers. I did see a post saying the ICSP header isn't always exact, but I've only measured one of my boards and haven't compared to the others. Perhaps it's between manufacturers as well.

I've also found a pinout that matches the board I've got exactly (shamelessly stolen from https://forum.arduino.cc/t/problem-pin-d13-in-mega2560-pro-version-mini/649168): 
[image]

Alarm-Siren, 2023-05-13:
Now that you mention it, I can see in the picture that the ICSP and D53(etc) clusters on the right aren't actually *quite* aligned with the main clusters top and bottom, so probably best to get some measurements of them too.

I took it as an assumption, but the distances between the pins within a cluster are 0.1"? Might be best to check that. Also about the USB, good to know its flush with the edge but I actually meant how deep is the connector itself.

For clarity, please see attached. I think with these 12 additional measurements should be able to do a really good footprint that matches your board exactly. [image]

P.S. that pinout diagram is much clearer than the ones on the website, good find!

BartoilPerso, 2023-05-14:
Might be the photo, or my measuring skills, but the ISCP and D53 clusters are almost aligned, close enough at least that a set of 2x wide headers would fit if it weren't for the mounting hole right there. At least on this particular board, the spacing is pretty darn close across those. Thanks so much for your help with this, this library is incredibly helpful!

1: 0.240"
2: 0.100"
3: 0.100"
4: 0.100"
5: 0.100"
6: 0.770
7: 0.100"
8: 0.100"
9: 0.410"
10: 0770"
11: 0.100"
12: 0.100"

Alarm-Siren, 2023-05-17:
There's a preliminary version of the new symbol & footprint available here:
https://github.com/Alarm-Siren/arduino-kicad-library/archive/refs/heads/4.1.0.zip

I'd appreciate if you could check them over, give me your feedback. In particular it'd be worth printing it out and comparing against the modules you have in your possession.

BartoilPerso, 2023-05-18:
I think the only measurement that's off is the location of the right side mounting hole and the ICSP headers. Based on the points you asked for, I was off by a hair on measurement 9 - that should be 0.400", and in the footprint it's currently 0.350". that means the mounting hole should be 0.600" from the center to the top edge of the board in this orientation. (I'm going by the center of the silkscreened circle, vs their drill position in this case) and the footprint has it at 0.620"

The other mounting hole, and the USB are perfect!

I did check two other boards as an afterthought, and there is a very slight difference in locations, but it's within 0.01", so I'd say for 99%  of those who use the footprint from this library, that'll be plenty accurate! (and the others probably aren't using KiCad.. or knockoff Arduino clones!)

Alarm-Siren, 2023-05-18:
I've pushed a new commit with updated positioning, hopefully I've got it right this time. Same link as before will give you the updated version to check against.

BartoilPerso, 2023-05-18:
That is sooo close! Just the ICSP header needs to be moved up by .030". It should be 0.400" between D30 and GND4 in the ICSP header, and from 5V2 to D52 it should be 0.300".
The mounting holes are perfect now though! Thanks again for your work on this!

Alarm-Siren, 2023-05-18:
How about now?

BartoilPerso, 2023-05-19:
Sorry for the delay! Everything looks awesome!

Alarm-Siren, 2023-05-19:
Excellent. Obviously you've got your development copy for now, but the finalised version (I'll do my own double-checks, too) will be included in the 4.1.0 release late this month.

victorproye, 2023-06-21:
man i jus found your library and its awesome. tank you those foot prints save my day.

Alarm-Siren, 2023-06-21:
> man i jus found your library and its awesome. tank you those foot prints save my day.

I'm glad you have found it useful :)

Alarm-Siren, 2023-07-15:
Finalised in #90

closed 2023-07-15 by Alarm-Siren