Alarm-Siren/arduino-kicad-library #87: No Edge.Cuts present, Courtyard issues

0xMihir, 2023-05-18:
In all of the footprints, there are no lines or shapes on the `Edge.Cuts` layer, meaning that KiCad does not generate the outline for the board. Also, the lines on B.Courtyard make it impossible to place footprints on the shield.

Alarm-Siren, 2023-05-19:
Regarding `Edge.Cuts`, there is not meant to be anything on that layer. These footprints are to allow you to create your own PCB to attach to an Arduino module; not everyone will necessarily want their PCB to be shaped like the Arduino it is connected to. If you actually want it to be the same shape, it's easy enough to trace the `Silkscreen` outline. (Also, actually the `Arduino_Nicla_Voice_Tile` footprint *does* have some geometry on the `Edge.Cuts` layer because a cut-out is necessary to accomodate the components on the back of the Nicla Voice module.)

Regarding `B.Courtyard` (or `F.Courtyard`, as it is on some footprints), again this is deliberate. This does not prevent you from placing your footprints wherever you want, it just means you get errors/warnings in the DRC checker. This is intentional, because I do not model the locations of all tall components underneath the Arduino module. Basically its meant to be a warning to the designer: "you've put some components in the potential danger zone, you should check your components' skylines aren't going to conflict with something on your chosen Arduino module".

I don't know which footprint you were attempting to use, but you should also note that Socket footprints have `F.Courtyard` because the footprint is intended for attaching the Arduino module *on top of* your custom PCB, so the danger zone is underneath the module, i.e. on the topside of your custom PCB. Shield footprints have `B.Courtyard` because they go underneath so the danger zone is on the back side of your custom PCB.

I will add notes to the readme about both these topics, so everyone is clear on the intention.

0xMihir, 2023-05-22:
Thanks for the help!