Alarm-Siren/arduino-kicad-library #89: mkr 1300 - pins orders

crepseS, 2023-06-10:
Hi  in the official documentation  https://docs.arduino.cc/static/56e4f5bef7037ff6d3e450cdfd327de1/ABX00017-full-pinout.pdf  

pins order is not the same as in this library therefore it can not be plugged to an actual arduino mkr 1300 board.

Alarm-Siren, 2023-06-10:
I have reviewed the MKR 1300 symbol and the MKR 1200 footprint (which is re-used for the MKR 1300 symbol), and I can confirm that the pinout exactly matches the one given in the document you have linked to (see below).

Are you able to be more specific what problem you are having?

[image]

crepseS, 2023-06-11:
the problem seems solved when i migrated the lib to kickad 7

crepseS, 2023-06-11:
see the image 
[image]

crepseS, 2023-06-11:
Now I understand , on pcb  the pins are ok but on schematic they look not in the same position .   I guess this should not be a problem.

Alarm-Siren, 2023-06-11:
The location of the pins on the symbol have no relation to the physical locations of the pin on the module. This is normal for all schematic components in Kicad.

By convention, "inputs" are on the left of the symbol, "outputs" are on the right, power in on the top and ground connections are on the bottom (note the present version of this library does not fully follow this convention, but my point is its totally normal for the two to be unrelated)

closed 2023-06-11 by Alarm-Siren