Alarm-Siren/arduino-kicad-library #101: Seems to work with KiCad 8.0

ScarletThe4EyedCuda, 2024-02-26:
Seem to work under KiCad 8.0... 

They move the "Plugin and Content Manager" to the main menu or the "Tools" menu.

All I had to do was click on Libraries
Scroll down to KiCad Library for Arduino Modules, Click install
Click on close once finished.
Restart KiCad and the symbols were there.  

I loaded up the Clone Meta 2560 Pro Socket and it works just fine.

Alarm-Siren, 2024-02-29:
Thank you for the info, I haven't had a chance to look at Kicad 8 compatibility yet, so its nice to know it shouldn't be too problematic. Maybe just some Readme updates for the instruction

gilescampbell, 2024-11-23:
Hi, sorry for a rather simple question. How would I go about installing this library on KiCad 8 for Mac? I've searched through the menu system but cannot find the KiCad Library for Arduino Modules listed anywhere. cheers, Giles