Alarm-Siren/arduino-kicad-library #104: the footprint of Clone_Mega2560_Pro_Socket is wrong

mmmx2000, 2024-09-17:
the footprint available for Clone_Mega2560_Pro_Socket is the wrong size, as can in the picture 
[image]

Alarm-Siren, 2024-09-17:
Hi @mmmx2000

I'm sorry but I do not follow what the issue is. The footprint should be correct for the non-official Meag2560 Pro. Can you be more specific about how it is wrong? 

To be clear, if you're looking for the official Arduino Mega2560, there is a different component in the library for that. The Arduino Mega2560 and Clone Mega2560 Pro are different components with different footprints.

mmmx2000, 2024-09-17:
understood 
thanks

ScarletThe4EyedCuda, 2024-09-17:
It is the correct size. I have used it on 3 projects with no issue.

-------- Original Message --------

closed 2024-09-17 by Alarm-Siren