Alarm-Siren/arduino-kicad-library #106: Arduino Giga R1 Wifi Shield Missing Middle Pins

cfpbiggs, 2025-01-19:
Very new to this, so please correct me if I am mistaken, but I am making a shield for the Giga R1 Wifi, and the pinout/footprint available through this library seems to be missing the 44 extra pins in the middle of the board. Worth noting is that I am using KiCAD 8, so it is possible this is some new-version issue.

Here's what comes through via KiCAD for me:
[image]

And here's an image of the Giga R1 Wifi from the Arduino website with the seemingly missing pins highlighted.
[image]

Any help would be greatly appreciated!

Alarm-Siren, 2025-01-19:
Hi @cfpbiggs 

I did not include those pins as per the [module's datasheet](https://docs.arduino.cc/resources/datasheets/ABX00063-datasheet.pdf) (specifically pages 3 and 9) those connectors are labelled as not being intended for general purpose use, but specifically for [Arducam style camera modules](https://thepihut.com/products/0-3mp-ov7675-20-pin-dvp-camera-module-for-arduino-giga-r1-wifi) that should plug straight in.

I also vaguely recall being unsure of the exact dimensions of the connectors, though looking at them now they look to be standard 0.1" pitch so not sure why I wasn't sure.

I could probably add a variant that includes these pins, but I'd have no way of double checking correct placement as there doesn't appear to be CAD files available for this particular module, and I don't think any other Arduino modules feature these connectors... Do you have one of these modules in your posession?

cfpbiggs, 2025-01-19:
Hi @Alarm-Siren,

Thank you for your quick response, and that makes a lot of sense. I would greatly appreciate a variant if you have the time to make it!
I do not have one of these modules in my possession yet, but I was planning on ordering one soon. Unless someone beats me to it, I can place that order today and provide some measurements once it arrives.

cfpbiggs, 2025-02-09:
Hi @Alarm-Siren,

Here are some basic dimensions, but let me know if there are particular lengths/measurements you need or that would be more useful!

[image]

I used a pair of calipers to get all of these measurements, so they might not be perfect, but I think it's the best I can do with the tools I have.

Alarm-Siren, 2025-02-16:
Thank you, I think that should be sufficient for me to work with. I apologise for the slowness of response, to be perfectly honest I am finding it difficult to find motivation to work on this repo at the moment - too much else going on in life! Nevertheless I shall try to put out a beta build for you to test shortly.

cfpbiggs, 2025-02-20:
Hi @Alarm-Siren,

There's no need to push yourself if you don't presently have the time! I would be very grateful for the beta build, but I can wait a good while and work on other projects.

Thanks again for your help!

Alarm-Siren, 2026-11-16:
New symbol and footprint variant added in [v4.2.1-beta](https://github.com/Alarm-Siren/arduino-kicad-library/releases/tag/v4.2.1-beta).

@cfpbiggs I appreciate its been a long time and you've probably moved on, but even so I would be greatly obliged if you could check it against your Giga R1 module to make sure everything is properly aligned.